mungewell/pico-timecode #1: Can a camera Mic port provide enough 'Phantom' power for RP2040?

mungewell, 2023-05-10:
Thought crossed my mind, can we power the Pico (or other board) from the 'phantom power' provided via camera's mic port?

If so, we could make 'pico-timecode' into a dongle to plug directly in. Once sync'ed it could remain attached to/power by camera.

There is this small variant of a Pico board, which can also run micro-Python....
https://www.pishop.ca/product/tiny-2040-2-mb/

mungewell, 2024-03-01:
Background reading/watching...
https://www.youtube.com/watch?v=3PH9jzRsb5E

Xander1125, 2024-03-24:
This is not ideal because any power down of the camera or disconnection of the cable would result in a desync of the timecode devices. Typically timecode allows the user to power the camera down multiple times during a shoot while remaining sync. This encourages all-day shooting for events like weddings or other big shoots.

Also, phantom power won't be provided by the built in mic port of many cameras, if not all. An xlr connection is typically needed, because the pins remain separate even when connecting or disconnecting the cable. A 3.5mm or TRS cable could short circuit if the user is connecting or do connecting the camera while phantom power is active.