mungewell/pico-timecode #5: Testing Rev-1 Hardware

mungewell, 2024-02-25:
I drew up a schematic and laid out a PCB for Rev-1 of the hardware, the audio interface is needed to amp/buffer the audio LTC signal into digital signals to/from the Pico.
[image]

This board is on order, and hopefully I will have 'hands on' in about a week...

The board is heavily dependent of the Pico footprint developed here, so a big thanks to that project:
https://github.com/ncarandini/KiCad-RP-Pico

The board is designed so that a Pico can be soldered directly (via castellated pins), or using pin headers. It's also zoned into 3 sections, where the battery/output and 'top of pico' can be removed to make a much smaller board. _There are Fit/No-Fit resistors to route the LTC output to the Right Channel of the input jack when using the smaller board._

I'm 100% certain I'll need to change the BOM, hopefully the PCB layout is workable. :-)

mungewell, 2024-02-25:
Some notes on component selection:

U1 is a fairly generic Dual Op-Amp, there are many substitutions possible, I ordered a few parts to try:

- LM358DR2G - https://www.digikey.ca/en/products/detail/onsemi/LM358DR2G/661123?s=N4IgTCBcDaIDIFkDMBWAHAEQEpgOIgF0BfIA
- LM2904S-13 - https://www.digikey.ca/en/products/detail/diodes-incorporated/LM2904S-13/5306172?s=N4IgTCBcDaIDIFkwE4AMAWAzgWgIwGYQBdAXyA
- RC4558IDR - https://www.digikey.ca/en/products/detail/texas-instruments/RC4558IDR/701928

U2 is more specialized, it is a programmable gain Amp controlled over a SPI-like interface. The intent is that Pico Timecode can have a controllable output level (or even turn it off).

- MCP6S91-E/SN - https://www.digikey.ca/en/products/detail/microchip-technology/mcp6s91-e-sn/716209

_The rest are 'jelly bean' and really component values can be substituted for whatever values you have available...._

mungewell, 2024-03-10:
Assembled the first Rev-1 board, a few component tweaks and it's functional... Woo Hoo!

salesmendesandre, 2024-05-10:
Have any adjustments been made to the hardware output files? I would be interested in ordering some assembled units from jlcpcb. Is the gerbers_2sides_pico.zip and pt.csv files sufficient or is some modification necessary?

mungewell, 2024-05-15:
The only major mistake is not having a ChipSelect for the Programable Amp, which I fly wired across to Pin 5. Other than that the board works OK, although this amp doesn't quite have enough difference in gain (between mic and line level).

I built the two layer board as, in the end, there was no difference in the cost between 1 and 2 layers. I have excess boards, which I could possibly share....

salesmendesandre, 2024-06-10:
Hello,

Today the boards I ordered arrived (when you responded I had already placed the order on PCBWay).

Regarding the components, I am using the ones you indicated in the CSV file. U1 -> LM358, U2 -> MCP6S91-E/SN. In the previous comment, you mentioned that the biggest issue would be not being able to control the ChipSelect of U2 from the microcontroller. As it is connected to GND, should it work without problems as it is currently connected, or is any modification necessary?

I am using the latest code found in the main branch of the repository. Sometimes the display shows an underflow error. Is it correct to use the latest version found in the repository? I have not yet been able to get a jam between two (I will keep trying in the coming days to see if there is any issue from when I soldered the components).

[image]

One more question: R12 and R7 are not necessary to mount, right? And R9 and R11 are soldered directly, correct?

Thank you very much for everything.

mungewell, 2024-06-13:
Nice, always liked black solder resist... though you should have gone with yellow silkscreen ;-)

Correct R7/R12 are not-fit, they are options to route audio/LTC if/when you only have the one connector. R11 is for 'bridging' the audio (say on camera mic) from J1 to J2.

R9 is a provision for tweaking signal from external LTC through AMP, this need to cross the reference voltage created with U1B pin 7.

Did you find lift the pin for the CS? Here's how I wired it across to GPIO5 of the Pico
[image]

You'll also want to be running this branch:
https://github.com/mungewell/pico-timecode/tree/pcb_rev1

I am a little distracted with other projects, but do intend to come back this one and validate those alternate XTALs.

salesmendesandre, 2024-06-14:
I have connected the CS pin to GPIO5, and now the output has TIMECODE LCT (I tested it with the application: https://timecodesync.com/monitor/).

I noticed there are differences between the resistors you are using compared to those listed in the CSV file along with the Gerber:

mungewell, 2024-06-15:
The schematic as 10uF, I think that 10nF might be a bit too small.
https://github.com/mungewell/pico-timecode/blob/main/hardware/output/schematic.pdf

The resistors aren't really critical, what you want to do is check that incoming LTC is causing the output of U1/1 is toggling, and this is a large enough swing that the PICO sees it as changing input.

For LTC output the programable amp doesn't have enough swing to change between Line and Mic level. The R5/R6 scale the signal down, and then the amp amplifies.... if you are not getting enough output for 2nd device to sync check the menus that the level is set as high as possible.

Do you have something that could scope the signals? Even a DVM would be helpful to check the DC levels of the reference/etc.

mungewell, 2024-06-15:
IIRC one of the amps I tried in the U1 position had difficult getting close to it's rails... I think I worked around this by adding a larger (330ohm???) at R9 to drop the signal more so that the Pico could 'see' it.