mungewell/pico-timecode #6: Suggestions for Rev-2 Hardware

mungewell, 2024-03-10:
As development moves forward, I'll 'open the floor' to suggestions from every/anyone.

The following are on my list:

- Button footprints for displays without buttons
- Connections for I2C 7-seg display and switches, to assemble as Digi-Slate
- Battery holder, connector footprint and diode footprint
- Diode footprints for 'Bias-Power' (from camera audio input). 

(I'll update list as suggestions come in).

salesmendesandre, 2024-03-10:
First of all, I would like to congratulate you for the development you are doing. 

The device you are developing is designed to be used in cameras/recorders that have two audio channels (for example the right one for the microphone audio and the left one for the LTC).

I would like to propose you to include the possibility of including the necessary hardware so that it can be used in single channel devices without the need to disconnect the device and connect the microphone. In a single channel I suppose it would be possible to use the LTC at the beginning of the audio (x seconds) and then the normal audio. I don't know how this could be solved at the hardware level, but something that would change the output of the ltc to a direct connection to an input jack.

Once again I would like to congratulate you for the work you are doing.

mungewell, 2024-03-10:
Pass-though audio is supported now, if assembled in the construction with two 3.5mm connector. LTC output is on Tip/Left, Ring/Right is connected through to input connector (ie Mic).

I was tinkering with audio levels today and grabbed this screen shot (running in the 'Mic' output level, with Rode Video Mic attached to input). 
[image]

I am not 100% sure what 'Bias' levels various camera/recorders output. My Sony cameras seem to out 2.7V on Ring (or TRS), my Tascam audio recorder output 2.2V on Ring. 

Are there really devices which can't cope with stereo audio (even if external Mic is only mono)? Seems to me that switching would be a pain, and likely screwed up by operator on set.

As to electronically switching output between Mic and LTC; this might be possible if MCP6S93 was used, but it is a little bit more expensive ($2.21 vs $1.79).

Xander1125, 2024-03-24:
Although I don't think may would use the feature, the request above is for a toggle button that switches between mono pass through and LTC at the click of a button, so the user could start a clip with 10 seconds of timecode, then hit the button and use their mic without having to disconnect the LTC device from the signal chain.

I don't think many people will need this feature though, I've never seen a camera (at the level where someone is considering using a timecode workflow) with mono input only.