mungewell/pico-timecode #7: Direct connection to Sony Multi-Interface Shoe?

mungewell, 2024-03-12:
I have a couple of Sony cameras without an audio jack, so I was wondering.... and ended up going down a massive rabbit hole.

The Multi-Shoe is described:
https://en.wikipedia.org/wiki/Multi_Interface_Shoe

It seems that (at least for simple devices) that the function is 'encoded' on the first 3 pins, as pull up/downs. It looks like for analogue audio that there are connections on:

- pin 4 - Mic GND
- pin 5 - Mic Right
- pin 6 - Mic Left

also

- pin 7/15/21 - Power GND
- pin 13 - Regulated power (3.1V)
- pin 14 - Mic Center(?)
- pin 17 - Un-reg Power (6.1-8.4V)

There are also some Digital Mic, with a digital connection to the camera(s), I presume that's the `GVIF_SDATA_N` / `GVIF_SDATA_P` alternate signals on the same pins.
 
There is a database of mics, and what they are supported on:
https://support.d-imaging.sony.co.jp/www/cscs/accessories/products.php?lang=en&area=gb&mnt=29

The real question is whether there's a cheap/easy way to get a compatible connector? Can we re-arrange the pins on a 
Sony ADPMAA adapter?
[image]

mungewell, 2024-03-12:
There's another description here:
https://www.mhohner.de/sony-minolta/flashcomp_techref.php

And Sony make an adapter for some wireless kit:
https://www.bhphotovideo.com/c/product/1272753-REG/sony_smad_p3d_smadp3d_multi_interface_shoe_adapter.html

jyz0501, 2024-04-07:
神牛领夹麦和大疆麦克风 2 转换头已经有两款产品了

mungewell, 2024-10-27:
It also appears that the (older) Nex-3/5 also supported a hot shoe mic, but the connection is different (vertical plan, rather than horizontal).
https://support.d-imaging.sony.co.jp/www/cscs/accessories/compatibility.php?area=gb&lang=en&mdl=ECM-SST1&cat=3

Multi-shoe replacement connectors intended for repairing flash guns are now appearing on eBay for around $15. This might be hackable, to move pins to places we need them...
[image]

mungewell, 2024-10-27:
The XLR-K1M mic adapter has a flying lead connection to the hot-shoe, which is available as a spare part - though kind of expensive....
https://sonyparts.com/products/1-846-330-11?srsltid=AfmBOopgottERDM1yLpO9kUrYRDZwedYgeac7rRLAyB7CacxPdOG2-0m

mungewell, 2024-10-27:
There's also an extension cable for use with the XLR-K3M:
https://sony.encompass.com/item/12536372/Sony/1-002-838-11/

mungewell, 2024-10-27:
And replacement shoe for the XLR-K2M:
https://sony.encompass.com/item/10344058/Sony/A-2063-272-A/

mungewell, 2024-10-27:
This is the hot-shoe connector from the FX-3's TopHandle, XLRH1:
https://sony.encompass.com/item/12636514/Sony/A-5030-779-A/

https://www.partstown.com/sony/sonya-5030-779-a?srsltid=AfmBOoraCnVOq-5dJzyZLbZ0TQF_aqHCmNoBujVAVY2WT-ynjtW0YPb7

mungewell, 2024-11-09:
"Multi-shoe replacement connectors intended for repairing flash guns", I got some to experiment....

Firstly of note is that the connections are made via the FFC, which is inserted vertically into the body of the 'connector'. The wire pins of the MI seem to be bent at 90' (haven't disassembled yet) and make contact with pads off the FFC.
[image]
[image]

[image]

The pads which are Pins 1..3 on the MI are hard wired to (assumed) Pin 7, which would be ID 0b000. Assuming the ID pins are short to ground or open, the whole table would be:
```
Pin1	Pin2	Pin3	Meaning
ID3	ID2	ID1
G	G	G	Flash
G	G	O
G	O	G
G	O	O
O	G	G
O	G	O
O	O	G
O	O	O
```

The center portion of the connector is used for a floating pin (to hot-shoe center) which would block the pins that we'd want to connect to, but perhaps this can be modified. Maybe someone could 3D print a replacement part.

At least 'we' should be able to use this connection to probe out the possible IDs, I'll update the table if I figure anything out.

mungewell, 2024-11-28:
There's also this DJI adapter:
https://www.bhphotovideo.com/c/product/1816274-REG/dji_cp_rn_00000332_01_mic_2_camera_adapter.html

dylanreeve, 2025-08-11:
This AliExpress seller is listing a replacement shoe for the XLR-H1 audio adapter. It appears to be a full MI-pinned show. The XLR-H1 is a dual channel XLR input. 

https://www.aliexpress.com/item/1005008851735668.html

mungewell, 2025-12-28:
This appears to be the official part (that AliExpress is offering).
https://sony.encompass.com/item/12636514/Sony/A-5030-779-A/

_No good pictures of the reverse side though_

mungewell, 2026-02-07:
The discount SMAD-P3 arrived. 

On my HX60 it popped up the message say device was not supported (without receiver attached - I don't have one). It did not present the message when the flex was removed from the PCB, suggesting that ID is asserted by the PCB (not the flex).

Plastic casing is easily removed with 3x screws on case, and 4x long screw in the 'strut' which holds the MI connect head. The MI head is fully populated, and has a 21way connection to that end of the flex. Looks like the flex provides all the connections that the Pico-Timecode device would need...

[image]

[SMAD-P3.zip](https://github.com/user-attachments/files/25141697/SMAD-P3.zip)

psiegl, 2026-08-28:
I am also keen to understand, if Sony MIs 3 IDs can be reverse-engineered properly and thus one could create hotshoes.
For this matter, I attempted @mungewell approach, pushed the libRsrcHotshoe.so into ghidra  (outcome: [libRsrcHotshoe.so.ghidra.c](https://github.com/user-attachments/files/31567507/libRsrcHotshoe.so.ghidra.c)) and let Ghidras output be interpreted by Anthropics Claude (outcome: [libRsrcHotshoe.so.claude-ghidra-interpreted.c](https://github.com/user-attachments/files/31567508/libRsrcHotshoe.so.claude-ghidra-interpreted.c)).

Furthermore, I tried to push as much information as given about MI, AAS, iISO and in particular SAT / SAT2 into Claude / ChatGPT. My current understanding:

 ## Sources and references

- **Sony MI Shoe reverse engineering / core pinout**  
  Forum archive with the MI Shoe pinout, SAT/SAT2 and Minolta/Sony AAS/iISO relationship, the reported 3.15 V `LOGIC_VDD` rail, the `UNREG` battery rail, and accessory-ID observations:  
  https://www.so-fo.de/t32344f291-Sonys-neuer-Multi-Interface-Shoe-MIS-2.html#msg292003

- **Sony ADP-MAC teardown**  
  MI-foot-to-Active-Interface-Shoe adapter mapping; notably `ID3 = 150 kΩ → GND`, pass-through `ID1`/`ID2`, and `UNREG` routed to the AIS `SHOE_UNREG` contacts:  
  https://www.so-fo.de/t34245f291-Und-noch-ein-Adapter-Sony-ADP-MAC-AIS-auf-MIS.html

- **Sony ADP-AMA / ADP-MAA teardowns**  
  Passive flash-adapter mappings; `ADP-MAA` ID profile (`ID3 = 470 kΩ → GND`, `ID2 = GND`, `ID1 = GND`), flash-signal routes, and absence of `LOGIC_VDD` / `UNREG` routing:  
  https://www.so-fo.de/t33060f291-Innenleben-Sony-Flash-Adapter-ADP-AMA-und-ADP-MAA.html#msg293602

- **GitHub: pico-timecode, issue #7 — “Direct connection to Sony Multi-Interface Shoe? (this discussion)”**  
  Replacement MI-foot observations, including feet with `ID3`, `ID2`, and `ID1` tied to ground (`GGG`) and practical MI Shoe pinout discussion:  
  https://github.com/mungewell/pico-timecode/issues/7

- **DPReview: “Smart Accessory identification, need help”**  
  NEX-5R/SAT resistor-ID experiments. Reported profiles include SAT flash (`22 kΩ / 22 kΩ`), EVF (`~56 kΩ / ~56 kΩ`), and a reportedly working analogue microphone identity: `ID1 = 450–470 kΩ`, `ID2 = 120–150 kΩ`:  
  https://www.dpreview.com/forums/threads/smart-accessory-indentification-need-help.4101342/

- **Impulsite.ru: “Sony HVL-F7S — connecting an external flash to Sony NEX-3/NEX-5”**  
  NEX Smart Accessory Terminal / HVL-F7S analysis: `UNREG`, `PGND`, `LOGIC_VDD`, `CLK`, `SIO`, `STROBE`, `ID1`, `ID2`, and reported ~100 kΩ camera-side pull-ups on the ID lines:  
  https://impulsite.ru/viewtopic.php?p=7740

- **Sony ECM-W3 / ECM-W3S documentation**  
  MI Shoe digital-audio receiver documentation, including receiver power supplied from the camera via the MI Shoe:  
  https://helpguide.sony.net/ilc/2340/v1/en/contents/TP1001273671.html  
  https://helpguide.sony.net/ilc/2340/v1/en/contents/TP1001273675.html

- **Sony ECM-S1 documentation**  
  MI Shoe-powered receiver specifications: built-in battery or power supply through the MI Shoe, rated at 5 V:  
  https://helpguide.sony.net/ilc/2350/v1/en/contents/TP1001273745.html

- **Michael Hohner — “Technical references” / Minolta–Sony flash-compendium reference**  
  Consolidated technical reference for Minolta AF ISO, Minolta/Sony iISO / Auto-lock Accessory Shoe, and Sony Multi Interface Shoe pinouts. Includes the MI Shoe power pins (`LOGIC_VDD` 3.15 V, `UNREG` battery supply reportedly fused at 1.6 A, `REG_GND`, `PGND`), ID pins, flash/EVF/GPS/WLAN/audio multiplexing, the MI-to-legacy flash mapping, and the documented flash ID pattern (`ID1 = GND`, `ID2 = GND`, `ID3 = 470 kΩ → REG_GND`). It also documents the older Minolta Dynax/Maxxum 3000i / Alpha 3700i power contacts: `F5/VDD1` (+5 V regulated), `F7/PGND`, and `F8/VDD0` (+6 V unregulated).  
  https://www.mhohner.de/sony-minolta/flashcomp_techref.php

- **Firmware reverse engineering: `libRsrcHotshoe.so`, Sony DSC-HX60**  
  Local Ghidra/Claude analysis: six-level calibrated ADC ID ladder; default threshold codes `904`, `712`, `508`, `308`, `68`, `0`; NVRAM IDs `0x16c0003`–`0x16c0008`; 3-byte Shoe-ID-to-GPIO-profile registry in the `BIN! ... FIN!` NVRAM blob at `0x1120005`; and two-consecutive-poll debounce with a nominal 100 ms poll interval.

## Sony Multi Interface Shoe: current reverse-engineering summary

### Executive summary

The Sony Multi Interface Shoe (MI Shoe / MIS) is a multiplexed, camera-managed accessory interface. It combines:

- the older Minolta/Sony **Auto-lock Accessory Shoe** (`AAS`, commonly called `iISO`) flash interface;
- Sony NEX **Smart Accessory Terminal** (`SAT` / `SAT2`) functions;
- MI-specific contacts and an additional third ID input.

It is **not** an always-powered connector. The body reads accessory-identification voltages, classifies the accessory, and then enables only the relevant rail and signal subset.

The most likely route to obtain `UNREG` and possibly `LOGIC_VDD` is therefore to present an ID tuple recognized as a **powered SAT/AIS-class accessory**, not a flash adapter identity.

---

## Relevant MI Shoe contacts

| MI pin | Signal | Notes |
|---:|---|---|
| 1 | `ID3` | Third MI-specific accessory-identification input |
| 2 | `ID2` | Accessory ID; corresponds to SAT `ID2` |
| 3 | `ID1` | Accessory ID; corresponds to SAT `ID1` |
[... 156 lines cut ...]
With a ~100 kΩ camera pull-up, a 470 kΩ pull-down is not electrically equivalent to open.

---

[... cut ...]
| `FA-CS1M` | Not connected | Not connected | `REG_GND` and `PGND` passed independently | MI flash extension |
| `ADP-MAC` | Not connected | Routed to AIS `SHOE_UNREG` | Adapter-specific ground links | MI body → Active Interface Shoe powered accessory |

Consequences:

```text
Flash profile ≠ useful general-purpose power profile.
```

The `ADP-MAA`, `ADP-AMA`, and `FA-CS1M` families are intentionally flash-only and do not carry either MI power rail.

```text
ADP-MAC / SAT / AIS accessory path
= strongest historical candidate for accessing UNREG.
```

psiegl, 2026-08-28:
# Sony MI Shoe: α6600 Confirmation Summary

## What's New

The α6600 analysis **confirms all HX60 findings**—same six-level ADC thresholds (`904/712/508/308/68/0`), same quantization, same 3-byte Shoe-ID mechanism with `0x1120005` profile table.

A second Sony `libRsrcHotshoe.so` binary, this time from a **Sony α6600 (ILCE-6600)** running **System Software / Firmware v1.10**, was imported into Ghidra ([libRsrcHotshoe.so.ilce6600.c](https://github.com/user-attachments/files/31575876/libRsrcHotshoe.so.ilce6600.c)) and compared with the previously analysed DSC-HX60 implementation.

The firmware was sourced from Sony's official support site: https://www.sony.de/electronics/support/e-mount-body-ilce-6000-series/ilce-6600/downloads/00016897

---

## Key Additions from α6600

### 1. Two- or Three-Channel ADC Configuration

The firmware contains an **explicit configuration flag** that determines whether the system uses **two or three ADC channels** for Shoe-ID generation.

**Implication:** Not all cameras necessarily have three physical analogue ID inputs. The presence of a 3-byte Shoe ID in the code does not prove every camera variant has three independent voltage dividers. Some models may use only two channels, with the third byte possibly defaulted or ignored.

This explains why some accessories might work across camera generations while others do not—the number of ID pins physically present could vary.

---

### 2. Explicit Special-Case Handling

The firmware contains **dedicated code paths** for at least two specific Shoe-ID states, rather than treating them as ordinary profile lookups:

| State | Meaning | Voltage (approx. @ 3.15 V) |
|-------|---------|---------------------------|
| `FF FF FF` | Invalid/unresolved—ADC sentinel path, not a real level | N/A |
| `01 01 01` | All three channels in level 1 (above highest threshold) | Each channel > 2.78 V |

**Why this matters:**

- `FF FF FF` confirms there is a **distinct "no accessory" or "fault" state** that is not simply another ID combination.
- `01 01 01` receiving special treatment suggests this state may represent a **specific high-voltage condition**—possibly a **passive accessory** that pulls all ID lines high, or a **default/unloaded** state that is handled differently from other level-1 combinations like `01 01 02`.

This provides a concrete reference point: if you measure all three ID pins above 2.78 V, the camera treats that as a meaningful special case, not just "ID 1,1,1".

---

### 3. Cross-Generation Confirmation

The same threshold ladder and architecture appearing in **two different Sony camera generations**:
- **DSC-HX60**: compact zoom camera
- **α6600 (ILCE-6600)**: APS-C interchangeable-lens camera, firmware v1.10

This strongly indicates this is a **reusable Sony platform implementation**, not a quirk of a single model, and increases confidence that the mechanism applies broadly across Sony's MI Shoe ecosystem.

---

## Still Unknown

- Actual accessory ID set (requires dumping `0x1120005` table from a real camera)
- 3.15 V ADC reference assumption (not confirmed in firmware)
- Semantic meaning of GPIO operations in profile table

psiegl, 2026-08-28:
# MI Shoe Reverse Engineering: NEX-6 v1.03 Confirmation

A third Sony `libRsrcHotshoe.so` binary has been analyzed—this time from the **NEX-6 (v1.03 firmware)**. The NEX-6 was the **first camera with the MI Shoe**, making this a critical data point for understanding how the system evolved ([libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31576414/libRsrcHotshoe.so.c)).

## What This Confirms

### 1. Same NVRAM IDs and Thresholds

| NVRAM ID | Default | NEX-6 | HX60 (v2.10) | α6600 (v1.10) |
|---|---|---|---|---|
| `0x16c0003` | 904 (`0x388`) | ✅ | ✅ | ✅ |
| `0x16c0004` | 712 (`0x2c8`) | ✅ | ✅ | ✅ |
| `0x16c0005` | 508 (`0x1fc`) | ✅ | ✅ | ✅ |
| `0x16c0006` | 308 (`0x134`) | ✅ | ✅ | ✅ |
| `0x16c0007` | 68 (`0x44`) | ✅ | ✅ | ✅ |
| `0x16c0008` | 0 | ✅ | ✅ | ✅ |

The same six-level ADC ladder has now been confirmed across **three different Sony generations** (NEX-6 → HX60 → α6600).

---

### 2. Same Architecture

| Feature | NEX-6 | HX60 | α6600 |
|---|---|---|---|
| Six ADC thresholds (`904/712/508/308/68/0`) | ✅ | ✅ | ✅ |
| 2/3 channel config (`g_variant_selector == 0x30`) | ✅ | ✅ | ✅ |
| `0x1120005` profile table (`BIN!`/`FIN!`) | ✅ | ✅ | ✅ |
| Debounce (2 polls) | ✅ | ✅ | ✅ |
| Quantization → 0xFF sentinel | ✅ | ✅ | ✅ |
| `/dev/gpio` interface | ✅ | ✅ | ✅ |

---

### 3. Key Difference: `01 01 01` Special-Case

The NEX-6 firmware **does NOT** contain the explicit `01 01 01` special-case handling found in the α6600.

| Function | NEX-6 | HX60 | α6600 |
|---|---|---|---|
| `01 01 01` special check | ❌ | ❌ | ✅ |
| `FF FF FF` special check | ❌ | ❌ | ✅ |
| UART/digital audio support | ❌ | ❌ | ✅ |

This indicates an evolution:

```text
NEX-6 (first MI Shoe):
    01 01 01 = normal ID → profile table lookup

HX60 (mid-generation):
    01 01 01 = normal ID → profile table lookup

α6600 (later generation):
    01 01 01 = special "no accessory" state → NOT profile table lookup

mungewell, 2026-09-06:
@psiegl Wow, lots of information there.... thank you!

As mentioned before the SMAD-P3 adapters (without receivers) pop up a message on the HX-60 saying 'device not supported'. If the 10way flex is disconnected (internally) they do not show this message.

- ID3 (pin 1) = 0 ohm to GND
- ID2 (pin 2) = 0 ohm to GND
- ID1 (pin 3) = ~89K to GND (though slowly changes to this so perhaps some capacitance/load is pulling current)

Also the last pin (24??) is 0 ohm to GND, see later comments.

I used your information to modify one of the flash bases I had brought, some horrible soldering and I have ID3 (pin 1) with 440K to ground as that's what resistors I had available. The HX-60 did not recognize any adapter (either pre or post mod), I suspect that there's another connection that it required.

The flex for these flash bases also brings pins 21..24 up to the internal connection (end held with tweezers), which doesn't connect to anything on my set up. I suspect that one (or more) of these needs to be connected to signal the presence of an adapter - ie before the IDs are evaluated.

Maybe I'll try the pin 24, and tie to ground?

[image]

mungewell, 2026-09-06:
NV-RAM-IDs; I wasn't sure whether you had reversed from the FW or read from a real camera.... I had done a full dump from HX-60 a while ago and confirmed that these are the same.

These are read/writable values, so if needed we could change them on the camera....
```
  
id=0x016c0003, size=0x0002, attr=0x00:
  88 03                                            ..
  
id=0x016c0004, size=0x0002, attr=0x00:
  c8 02                                            ..
  
id=0x016c0005, size=0x0002, attr=0x00:
  fc 01                                            ..
  
id=0x016c0006, size=0x0002, attr=0x00:
  34 01                                            4.
  
id=0x016c0007, size=0x0002, attr=0x00:
  44 00                                            D.
  
id=0x016c0008, size=0x0002, attr=0x00:
  00 00
```

Also dumped a few other cameras....
```
 grep -A 2 id=0x016c0003 *.bin.txt
Backup_a5000.bin.txt:id=0x016c0003, size=0x0002, attr=0x00:
Backup_a5000.bin.txt-  00 00                                            ..
Backup_a5000.bin.txt-
--
Backup_a5100.bin.txt:id=0x016c0003, size=0x0002, attr=0x00:
Backup_a5100.bin.txt-  00 00                                            ..
Backup_a5100.bin.txt-
--
Backup_a7Rmk2.bin.txt:id=0x016c0003, size=0x0002, attr=0x00:
Backup_a7Rmk2.bin.txt-  88 03                                            ..
Backup_a7Rmk2.bin.txt-
--
Backup_as100.bin.txt:id=0x016c0003, size=0x0002, attr=0x00:
Backup_as100.bin.txt-  00 00                                            ..
Backup_as100.bin.txt-
--
Backup_hx60.bin.txt:id=0x016c0003, size=0x0002, attr=0x00:
Backup_hx60.bin.txt-  88 03                                            ..
Backup_hx60.bin.txt-
--
Backup_nex6.bin.txt:id=0x016c0003, size=0x0002, attr=0x00:
Backup_nex6.bin.txt-  88 03                                            ..
Backup_nex6.bin.txt-
--
Backup_qx10.bin.txt:id=0x016c0003, size=0x0002, attr=0x00:
Backup_qx10.bin.txt-  00 00                                            ..
Backup_qx10.bin.txt-
```

mungewell, 2026-09-06:
Do we know what the 'profile table' actually means, can we decode it to match the HX-60's stated/supported accessories?
```
id=0x01120005, size=0x0100, attr=0x00:
  42 49 4e 21 02 05 08 00 00 02 11 00 05 06 06 00  BIN!............
  01 06 06 06 00 00 06 06 02 00 00 03 04 06 01 00  ................
  06 02 06 00 00 46 49 4e 21 00 00 00 00 00 00 00  .....FIN!.......
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

mungewell, 2026-09-06:
```
id=0x01120005, size=0x0100, attr=0x00:
42 49 4e 21 = BIN!

02 = items 
05 = data length?

loop1:
08 00 00
02 11 00

loop2:
05 06 06 : 00, 01 
06 06 06 : 00, 00
06 06 02 : 00, 00 
03 04 06 : 01, 00
06 02 06 : 00, 00

46 49 4e 21 = FIN!
```

Maybe loop1 describes the type of supported device, and loop2 matches the IDs to those device?

mungewell, 2026-09-06:
In case people don't know that they can connect to and read/write the camera's database... took a bit of futzing with Python libs, but it does (still) work.
```
$ sudo python3 pmca-console.py updatershell
No native drivers available
Using drivers libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-HX60W

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
100%
Starting updater shell...

Welcome to platform shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>bk r 0x01120005
42 49 4e 21 02 05 08 00 00 02 11 00 05 06 06 00 01 06 06 06 00 00 06 06 02 00 00 03 04 06 01 00 06 02 06 00 00 46 49 4e 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
>exit
Done
```

mungewell, 2026-09-06:
As mentioned above we can **WRITE** to the camera's database, so can change this '0x01120005' table to match the devices we actually have to hand...

Using @psiegl  table

| Level | Practical pull-down to `REG_GND` | Approx. ID voltage |
|---|---|---:|
| L1 | Open | ~3.15 V |
| L2 | 470 kΩ | ~2.59 V |
| L3 | 120–150 kΩ | ~1.72–1.89 V |
| L4 | 47–56 kΩ | ~1.01–1.13 V |
| L5 | 22 kΩ | ~0.57 V |
| L6 | Direct GND / a few ohms | ~0 V |

My SMAD-P3 adapter has:

-   ID3 (pin 1) = 0 ohm to GND -> L6
-   ID2 (pin 2) = 0 ohm to GND -> L6
-   ID1 (pin 3) = ~89K to GND -> L3 or L4?

So for the HX-60 assuming 'item1' is the Mic driver, maybe we want to change '05 06 06' to '04 06 06'?

BTW I think 'item2'  (ID '03 04 06') may be the **FDA-EV1MK** viewfinder.
https://support.d-imaging.sony.co.jp/www/cscs/accessories/compatibility.php?area=gb&lang=en&mdl=FDA-EV1MK&cat=1

mungewell, 2026-09-06:
Quick script to decode the table

[sony_mi.py.txt](https://github.com/user-attachments/files/31885407/sony_mi.py.txt)

```
$ python3 sony_mi.py
Container: 
    driver = 2
    id = 5
    drivers = ListContainer: 
        b'\x08\x00\x00' (total 3)
        b'\x02\x11\x00' (total 3)
    ids = ListContainer: 
        Container: 
            id = b'\x05\x06\x06' (total 3)
            driver = ListContainer: 
                0
                1
        Container: 
            id = b'\x06\x06\x06' (total 3)
            driver = ListContainer: 
                0
                0
        Container: 
            id = b'\x06\x06\x02' (total 3)
            driver = ListContainer: 
                0
                0
        Container: 
            id = b'\x03\x04\x06' (total 3)
            driver = ListContainer: 
                1
                0
        Container: 
            id = b'\x06\x02\x06' (total 3)
            driver = ListContainer: 
                0
                0
```

psiegl, 2026-09-06:
Where to begin? I did quite some things... 

# Determining `libHotShoe.so` variance across different camera generations.

## Threshold ladder

> **Confirmed identical across seven firmware versions.**

| Level | NVRAM ID | Default (hex) | Default (dec) | ADC range for this level | Voltage range (10-bit ADC, \(V_\mathrm{ref} \approx 3.15\text{ V}\)) |
|---:|---|---:|---:|---:|---:|
| 1 (highest) | `0x16c0003` | `0x388` | 904 | 904–1023 | 2.78–3.15 V |
| 2 | `0x16c0004` | `0x2c8` | 712 | 712–903 | 2.19–2.78 V |
| 3 | `0x16c0005` | `0x1fc` | 508 | 508–711 | 1.56–2.19 V |
| 4 | `0x16c0006` | `0x134` | 308 | 308–507 | 0.95–1.56 V |
| 5 | `0x16c0007` | `0x44` | 68 | 68–307 | 0.21–0.95 V |
| 6 (lowest) | `0x16c0008` | `0x0` | 0 | 0–67 | 0–0.21 V |

> [!NOTE]
> The voltage conversion assumes `LOGIC_VDD = 3.15 V`, based on the previously noted value. This has not been independently confirmed and should be validated empirically.

## Cross-firmware comparison

| Camera | SW Version | Ghidra |Segment | Approx. year | Threshold table | Result |
|---|---|---|---|---:|---|---|
| NEX-6 | v1.03 | [FirmwareData_NEX6V103.dat_fwtool_libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31885540/FirmwareData_NEX6V103.dat_fwtool_libRsrcHotshoe.so.c) | APS-C mirrorless; unstripped symbols | 2012 | Same algorithm and constants | ✅ Functional match; different build |
| DSC-HX60 | v2.10 | [FirmwareData_DSCHX60VV210.dat_fwtool_libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31885546/FirmwareData_DSCHX60VV210.dat_fwtool_libRsrcHotshoe.so.c) | Compact | 2014 | `904, 712, 508, 308, 68, 0` | ✅ Baseline |
| ILCE-7M2 (A7 II) | v4.01 | [FirmwareData_ILCE7M2V401.dat_fwtool_libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31885563/FirmwareData_ILCE7M2V401.dat_fwtool_libRsrcHotshoe.so.c) | Full-frame mirrorless | 2014 | `904, 712, 508, 308, 68, 0` | ✅ Match |
| ILCE-6000 (A6000) | v3.21 | [FirmwareData_ILCE6000V321.dat_fwtool_libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31885568/FirmwareData_ILCE6000V321.dat_fwtool_libRsrcHotshoe.so.c) | APS-C mirrorless | 2014 | `904, 712, 508, 308, 68, 0` | ✅ Match |
| DSC-RX100M4 | v2.00 | [FirmwareData_DSCRX100M4V200.dat_fwtool_libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31885569/FirmwareData_DSCRX100M4V200.dat_fwtool_libRsrcHotshoe.so.c) | Compact | 2015 | `904, 712, 508, 308, 68, 0` | ✅ Byte-identical `.so` to HX60V |
| ILCA-99M2 (A99 II) | v1.01 | [FirmwareData_ILCA99M2V101.dat_fwtool_libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31885572/FirmwareData_ILCA99M2V101.dat_fwtool_libRsrcHotshoe.so.c) | Full-frame SLT flagship | 2016 | `904, 712, 508, 308, 68, 0` | ✅ Byte-identical `.so` to HX60V |
| ILCE-6600 (A6600) | v1.10 | [FirmwareData_ILCE6600V110.dat_fwtool_libRsrcHotshoe.so.c](https://github.com/user-attachments/files/31885580/FirmwareData_ILCE6600V110.dat_fwtool_libRsrcHotshoe.so.c) | APS-C mirrorless | 2019 | `904, 712, 508, 308, 68, 0` | ✅ Match |

Consequently, the `libHotShoe.so` logic stayed merely the same. Some small changes was identified, which I mentioned previously.

| Feature / Aspect | NEX-6 (V1.03) | HX60/α99II/α7II (Older Gen) | α6600 (V1.10) | ILCE-6000/α7M2/etc (Newer Gen) |
| :--- | :---: | :---: | :---: | :---: |
| `01 01 01` Special Check | ❌ | ❌ | ✅ | ❌ |
| `FF FF FF` Special Check | ❌ | ❌ | ✅ | ❌ |
| UART Control Logic | ❌ | ❌ | ✅ | ❌ |
| Explicit Debug Logging | ❌ | ❌ | ❌ | ✅ |
| `/dev/mem` Mapping | ❌ | ❌ | ✅ | ❌ |
| `vsnprintf` Usage | ❌ | ❌ | ❌ | ✅ |

## Backup ID Usage Across All Models

The threshold loading uses **TWO** different backup IDs:

### 1. Threshold Ladder (the 6 values: 904, 712, 508, 308, 68, 0)
- **Backup ID:** `0x16c0003` through `0x16c0008` (one ID per threshold)
- **This is the same across ALL models**

| Model | Threshold Array Location |
| :--- | :--- |
| NEX-6 (V1.03) | `DAT_00019f22` - `DAT_00019f2c` |
| HX60/α99II/α7II (Older Gen) | `DAT_0001ac7e` - `DAT_0001ac88` |
| α6600 (V1.10) | `DAT_0001b1fe` - `DAT_0001b208` |
| ILCE-6000/α7M2/etc (Newer Gen) | `DAT_0001de72` - `DAT_0001de7c` |

### 2. GPIO Configuration Table (for signal routing)
- **Backup ID:** `0x1120005`
- **This is also the same across ALL models**

### Complete Backup ID Reference

| Backup ID | Purpose | Size | All Models |
| :--- | :--- | :---: | :---: |
| `0x16c0000` | Hotshoe Type (0=no shoe, 0x30=shoe) | 4 bytes | ✅ |
| `0x16c0001` | Logic VDD GPIO active level | 1 byte | ✅ |
| `0x16c0002` | Poll interval time (ms) | 4 bytes | ✅ |
| `0x16c0003` - `0x16c0008` | ADC thresholds (6 values: 904, 712, 508, 308, 68, 0) | 2 bytes each | ✅ |
[... 227 lines cut ...]
| **UART Wait Time (0x16c0010)** | ❌ | ❌ | ❌ | ✅ | ❌ |

### α6600 UART State Machine Transitions

The α6600 implements a state machine for UART control based on Shoe ID transitions:

| Previous ID (gShoeId) | New ID (gShoeIdBk) | Action | Description |
|-----------------------|-------------------|--------|-------------|
| `FF FF FF` | Any valid ID | **Enable UART** | No accessory → Accessory detected |
| `FF FF FF` | `01 01 01` | **Enable UART** | No accessory → Special accessory |
| Any valid ID | `FF FF FF` | **Disable UART** | Accessory removed |
| Any valid ID | `01 01 01` | **Disable UART** (if GPIO allows) | Normal accessory → Special accessory |
| `01 01 01` | Any valid ID | **Disable UART** | Special accessory → Normal accessory |
| `01 01 01` | `FF FF FF` | **Disable UART** (if GPIO allows) | Special accessory → No accessory |
| Any valid ID → Any valid ID | - | **No UART change** | Normal accessory change |

#### Helper Functions

| Function | Purpose |
|----------|---------|
| `FUN_00012674()` | Checks if ID equals `FF FF FF` |
| `FUN_0001269c()` | Checks if ID equals `01 01 01` |
| `FUN_000126c4()` | Checks if ID is NOT `FF FF FF` AND NOT `01 01 01` |
| `FUN_0001265c()` | Resets ID to `FF FF FF` |
| `FUN_000126e6()` | Notifies new Shoe ID via `RsrcHotshoeNotifyShoeID()` |

psiegl, 2026-09-06:
Above is all what I managed to reverse-engineer for now. But I am stuck.

If I compare your data ( @mungewell ) with the Alpha 6700, and take into consideration the aspect of the threshold ladder, then I don' t know how to interpret the Shoe-IDs? I mean, the device which becomes attached does not need re-soldering a resistor for a certain type of camera ;-) thus, why don't the Shoe-IDs match? If the 'actions' is different ... fine, different cameras. But at least I would expect that the Shoe-IDs overlap.

```
HX60 serialized example:

42 49 4e 21          "BIN!"
02                   P = 2 items/descriptors
05                   N = 5 Shoe-ID profiles
  
08 00 00             item 0
02 11 00             item 1
  
05 06 06 00 01       profile: Shoe-ID 05,06,06; actions [00,01]
06 06 06 00 00       profile: Shoe-ID 06,06,06; actions [00,00]
06 06 02 00 00       profile: Shoe-ID 06,06,02; actions [00,00]
03 04 06 01 00       profile: Shoe-ID 03,04,06; actions [01,00]
06 02 06 00 00       profile: Shoe-ID 06,02,06; actions [00,00]

46 49 4e 21          "FIN!"

Serialized size:
10 + 3P + N(3+P)
= 10 + 3×2 + 5×(3+2)
= 41 bytes -> fine
```

vs

```
α6700 serialized example:
  
42 49 4e 21          "BIN!"
01                   P = 1 item/descriptor
04                   N = 4 Shoe-ID profiles

03 08 00             item 0
  
05 05 02 01          profile: Shoe-ID 05,05,02; action [01]
05 05 04 01          profile: Shoe-ID 05,05,04; action [01]
06 06 05 00          profile: Shoe-ID 06,06,05; action [00]
05 06 04 01          profile: Shoe-ID 05,06,04; action [01]

46 49 4e 21          "FIN!"

Serialized size:
10 + 3P + N(3+P)
= 10 + 3×1 + 4×(3+1)
= 29 bytes -> fine
```

... more camera examples would be great. Thus, thanks for the shout out in the other git repo!

psiegl, 2026-09-06:
@mungewell reworked python script:

```python
from construct import *

# Each profile contains:
# - 3-byte Shoe ID
# - P action bytes
Profile = Struct(
    "shoe_id" / Bytes(3),
    "actions" / Array(this._.action_count, Byte),
)

# Main table structure
HotshoeTable = Struct(
    Const(b"BIN!"),
    "action_count" / Int8ub,        # P = number of action bytes per profile
    "profile_count" / Int8ub,       # N = number of Shoe-ID profiles

    # Check: P ≤ 16 (from Ghidra: if (0x10 < P) error)
    Check(lambda ctx: ctx.action_count <= 16),

    # Loop 1: P descriptors (3 bytes each: Grp, Bit, DefaultCtrl)
    "descriptors" / Array(this.action_count, Bytes(3)),

    # Loop 2: N profiles (3-byte Shoe ID + P action bytes)
    "profiles" / Array(this.profile_count, Profile),

    # Check: Footer must be "FIN!" (from Ghidra: if (*ptr != 0x214e4946) error)
    Const(b"FIN!"),
)

# List of (name, data) tuples
examples = [
    ("RX100M4", b"\x42\x49\x4e\x21\x00\x00\x46\x49\x4e\x21"),  # P=0, N=0, just BIN! FIN!
    ("HX60", b"\x42\x49\x4e\x21\x02\x05\x08\x00\x00\x02\x11\x00\x05\x06\x06\x00\x01\x06\x06\x06\x00\x00\x06\x06\x02\x00\x00\x03\x04\x06\x01\x00\x06\x02\x06\x00\x00\x46\x49\x4e\x21"),
    ("α6700", b"\x42\x49\x4e\x21\x01\x04\x03\x08\x00\x05\x05\x02\x01\x05\x05\x04\x01\x06\x06\x05\x00\x05\x06\x04\x01\x46\x49\x4e\x21"),
]

for name, data in examples:
    print(f"\n=== {name} ===")
    try:
        result = HotshoeTable.parse(data)
        print(f"Action Count (P): {result.action_count}")
        print(f"Profile Count (N): {result.profile_count}")
        print(f"Descriptors: {[d.hex() for d in result.descriptors]}")
        print(f"Profiles:")
        for i, profile in enumerate(result.profiles):
            print(f"  [{i}] Shoe ID: {profile.shoe_id.hex()}, Actions: {list(profile.actions)}")
    except Exception as e:
        print(f"Error parsing: {e}")
```

results in:

```bash
=== RX100M4 ===
Action Count (P): 0
Profile Count (N): 0
Descriptors: []
Profiles:

=== HX60 ===
Action Count (P): 2
Profile Count (N): 5
Descriptors: ['080000', '021100']
Profiles:
  [0] Shoe ID: 050606, Actions: [0, 1]
  [1] Shoe ID: 060606, Actions: [0, 0]
  [2] Shoe ID: 060602, Actions: [0, 0]
  [3] Shoe ID: 030406, Actions: [1, 0]
  [4] Shoe ID: 060206, Actions: [0, 0]

=== α6700 ===
Action Count (P): 1
Profile Count (N): 4
Descriptors: ['030800']
Profiles:
  [0] Shoe ID: 050502, Actions: [1]
  [1] Shoe ID: 050504, Actions: [1]
  [2] Shoe ID: 060605, Actions: [0]
  [3] Shoe ID: 050604, Actions: [1]
```

psiegl, 2026-09-06:
Further analysis ... I digged into the GPIO:

```
undefined4 FUN_00010eb4(undefined1 param_1, undefined1 param_2, int param_3)
{
    // param_1 = Grp (from descriptor byte 0)
    // param_2 = Bit (from descriptor byte 1)
    // param_3 = action byte
    
    __fd = open("/dev/gpio", 2);
    memset(auStack_28, 0, 0x10);
    local_24 = param_1;  // Grp
    local_20 = param_2;  // Bit
    
    if (param_3 == 0) {
        local_1c = 0x11;  // ← What does 0x11 mean?
    }
    else if (param_3 == 1) {
        local_1c = 9;     // ← What does 9 mean?
    }
    else {
        if (param_3 != 0x10) {
            // Error
        }
        local_1c = 5;     // ← What does 5 mean?
    }
    
    iVar1 = ioctl(__fd, 0xc0104702, auStack_28);
    // 0xc0104702 = GPIO_SET_MODE? 
}
```

## GPIO Mode Analysis (HX60)

Based on the assumption that `1` = output and `0` = input:

### Descriptors:

Item 0: 08 00 00
Item 1: 02 11 00

### Profiles Decoded:

| Profile | Shoe ID | Actions | GPIO0 (Item 0) guess | GPIO1 (Item 1) guess | Hardware Match |
|---------|---------|---------|----------------|----------------|----------------|
| 0 | `05 06 06` (22K,0R,0R) | `[0, 1]` | **Input** | **Output** | Stereo microphone (needs power/audio enable) |
| 1 | `06 06 06` (0R,0R,0R) | `[0, 0]` | **Input** | **Input** | Dumb Flash (passive trigger) |
| 2 | `06 06 02` (0R,0R,470K) | `[0, 0]` | **Input** | **Input** | ADP-MAA (passive adapter) |
| 3 | `03 04 06` (150K,56K,0R) | `[1, 0]` | **Output** | **Input** | AIS mic variant (needs power) |
| 4 | `06 02 06` (0R,470K,0R) | `[0, 0]` | **Input** | **Input** | Unknown passive accessory |

### GPIO Mode Summary

| Action | GPIO Mode giess | Typical Use |
|--------|-----------|-------------|
| `0` | **Input** | Passive accessories (flash trigger, adapters) |
| `1` | **Output** | Active accessories (microphones, powered devices) |
| `0x10` | ? | Special mode (not used in HX60 profiles) |

### Key Observations

1. **Microphones need an output GPIO** (audio enable / power control)
2. **Flashes and adapters use input GPIOs** (passive connections)
3. **Different microphone variants use different GPIO patterns**:
   - Profile 0: GPIO0 = Input, GPIO1 = Output
   - Profile 3: GPIO0 = Output, GPIO1 = Input

## HX60 Descriptors analysis

```
Item 0: 08 00 00
Item 1: 02 11 00
```

| Descriptor | Grp | Bit | DefaultCtrl | GPIO Pin |
|------------|-----|-----|-------------|----------|
| Item 0 | `08` | `00` | `00` | GPIO group 8, bit 0 |
| Item 1 | `02` | `11` | `00` | GPIO group 2, bit 17 (0x11 = 17) |

## How Descriptors and Actions Work Together

When an accessory is detected, the camera:

1. Looks up the profile based on the Shoe ID (resistor values)
2. Reads the actions from the profile
3. For each action:
   - Takes the corresponding descriptor (same index)
   - Extracts Grp and Bit from the descriptor
   - Passes Grp, Bit, and Action to `FUN_00010eb4()`
4. The function sets that specific GPIO pin to the mode specified by Action

## Example: Profile 0 (Stereo Microphone)
Profile 0: Shoe ID = 05 06 06, Actions = [0, 1]

| Index | Descriptor | Action | What Happens |
|-------|------------|--------|--------------|
| 0 | `08 00 00` (Grp=8, Bit=0) | `0` | GPIO (8,0) set to Input |
| 1 | `02 11 00` (Grp=2, Bit=17) | `1` | GPIO (2,17) set to Output |

## Visual Flow

```
Shoe ID (resistors) → Profile Lookup → Actions + Descriptors → GPIO Control

Example:
05 06 06 → Profile 0 → Actions [0, 1]
                         Descriptors [08 00 00, 02 11 00]
                         ↓
                         GPIO(8,0) = Input
                         GPIO(2,17) = Output
```

psiegl, 2026-09-06:
So, this is a bit of an assumption, but potentially it is as follows:

```python
from construct import *

# Each profile contains:
# - 3-byte Shoe ID
# - P action bytes (GPIO mode: 0 = input, 1 = output, 0x10 = special)
Profile = Struct(
    "shoe_id" / Bytes(3),
    "actions" / Array(this._.gpio_count, Byte),
)

# Main table structure
HotshoeTable = Struct(
    Const(b"BIN!"),
    "gpio_count" / Int8ub,          # P = number of GPIO pins configured
    "profile_count" / Int8ub,       # N = number of Shoe-ID profiles

    # Check: P ≤ 16 (from Ghidra: if (0x10 < P) error)
    Check(lambda ctx: ctx.gpio_count <= 16),

    # Loop 1: P GPIO descriptors (Grp, Bit, DefaultCtrl)
    "gpios" / Array(this.gpio_count, Bytes(3)),

    # Loop 2: N profiles (3-byte Shoe ID + P action bytes)
    "profiles" / Array(this.profile_count, Profile),

    # Check: Footer must be "FIN!" (from Ghidra: if (*ptr != 0x214e4946) error)
    Const(b"FIN!"),
)

def decode_action(action):
    """Decode action byte to human-readable format"""
    if action == 0:
        return f"in [{action}]"
    elif action == 1:
        return f"out [{action}]"
    elif action == 0x10:
        return f"special [0x{action:02x}]"
    else:
        return f"0x{action:02x}"

# List of (name, data) tuples
examples = [
    ("RX100M4", b"\x42\x49\x4e\x21\x00\x00\x46\x49\x4e\x21"),
    ("a7R II", b"\x42\x49\x4e\x21\x00\x00\x46\x49\x4e\x21"),
    ("HX60", b"\x42\x49\x4e\x21\x02\x05\x08\x00\x00\x02\x11\x00\x05\x06\x06\x00\x01\x06\x06\x06\x00\x00\x06\x06\x02\x00\x00\x03\x04\x06\x01\x00\x06\x02\x06\x00\x00\x46\x49\x4e\x21"),
    ("α6700", b"\x42\x49\x4e\x21\x01\x04\x03\x08\x00\x05\x05\x02\x01\x05\x05\x04\x01\x06\x06\x05\x00\x05\x06\x04\x01\x46\x49\x4e\x21"),
]

for name, data in examples:
    print(f"\n=== {name} ===")
    try:
        result = HotshoeTable.parse(data)
        print(f"GPIO Count (P): {result.gpio_count}")
        print(f"Profile Count (N): {result.profile_count}")
        
        # Print GPIO pins with raw hex in brackets
        if result.gpio_count > 0:
            print("GPIO Pins:")
            for i, gpio in enumerate(result.gpios):
                grp = gpio[0]
                bit = gpio[1]
                default = gpio[2]
                raw_hex = gpio.hex()
                print(f"  GPIO{i}(Grp={grp}, Bit={bit}) = DefaultCtrl={default} [{raw_hex}]")
        else:
            print("GPIO Pins: (none)")
        
        # Print profiles with proper table formatting
        if result.profile_count > 0:
            # Build header
            header_parts = ["Profiles:", "Shoe ID"]
            for j in range(result.gpio_count):
                header_parts.append(f"GPIO{j}")
            
            # Determine column widths
            col_widths = []
            for j in range(len(header_parts)):
                if j == 0:
                    col_widths.append(10)  # Index column
                elif j == 1:
                    col_widths.append(12)  # Shoe ID column
                else:
                    col_widths.append(16)  # GPIO columns (wider for "in [0]" format)
            
            # Print header
            header_line = "  ".join([f"{header_parts[i]:<{col_widths[i]}}" for i in range(len(header_parts))])
            print(header_line)
            print("-" * (sum(col_widths) + len(col_widths) * 2))
            
            # Print profiles
            for i, profile in enumerate(result.profiles):
                shoe_id = profile.shoe_id.hex()
                row_parts = [f"[{i}]", shoe_id]
                for action in profile.actions:
                    mode = decode_action(action)
                    row_parts.append(mode)
                
                row_line = "  ".join([f"{row_parts[i]:<{col_widths[i]}}" for i in range(len(row_parts))])
                print(row_line)
        else:
            print("Profiles: (none)")
    except Exception as e:
        print(f"Error parsing: {e}")
```

Result:
```bash
=== RX100M4 ===
GPIO Count (P): 0
Profile Count (N): 0
GPIO Pins: (none)
Profiles: (none)

=== a7R II ===
GPIO Count (P): 0
Profile Count (N): 0
GPIO Pins: (none)
Profiles: (none)

=== HX60 ===
GPIO Count (P): 2
Profile Count (N): 5
GPIO Pins:
  GPIO0(Grp=8, Bit=0) = DefaultCtrl=0 [080000]
  GPIO1(Grp=2, Bit=17) = DefaultCtrl=0 [021100]
Profiles:   Shoe ID       GPIO0             GPIO1           
--------------------------------------------------------------
[0]         050606        in [0]            out [1]         
[1]         060606        in [0]            in [0]          
[2]         060602        in [0]            in [0]          
[3]         030406        out [1]           in [0]          
[4]         060206        in [0]            in [0]          

=== α6700 ===
GPIO Count (P): 1
Profile Count (N): 4
GPIO Pins:
  GPIO0(Grp=3, Bit=8) = DefaultCtrl=0 [030800]
Profiles:   Shoe ID       GPIO0           
--------------------------------------------
[0]         050502        out [1]         
[1]         050504        out [1]         
[2]         060605        in [0]          
[3]         050604        out [1]         
```

psiegl, 2026-09-06:
So my issue is, if we consider the table ("Known adapter and accessory profiles") I gathered above by scraping the internet :-D and condense it:

| Profile | ID3 (pin 1) | ID2 (pin 2) | ID1 (pin 3) | Evidence |
|---------|-------------|-------------|-------------|----------|
| Electrically open | Open | Open | Open | Test condition |
| Replacement MI flash foot | GND | GND | GND | Observed |
| Sony ADP-MAA | 470 kΩ | GND | GND | Direct teardown |
| Sony FA-CS1M | 470 kΩ | GND | GND | Direct teardown |
| Sony ADP-MAC base | 150 kΩ | Passed to AIS | Passed to AIS | Direct teardown |
| Older AIS microphone | 150 kΩ | 150 kΩ | 470 kΩ | Historical |
| Older AIS XLR adapter | 150 kΩ | 22 kΩ | GND | Historical |
| SAT flash experiment | N/A | 22 kΩ | 22 kΩ | NEX-5R observed |
| SAT EVF experiment | N/A | 56 kΩ | 56 kΩ | NEX-5R observed |
| SAT analogue microphone | N/A | 120-150 kΩ | 450-470 kΩ | NEX-5R reported |
| SAT unsupported result | N/A | 22 kΩ | GND | NEX-5R reported |

Combined Mapping Table:

| Shoe ID Byte | Resistor | Confirmed? | Evidence |
|--------------|----------|------------|----------|
| `01` | Open? | ⚠️ Unconfirmed | Not yet observed in known profiles |
| `02` | 470K | ✅ Confirmed | ADP-MAA (470K,GND,GND → `06 06 02`) |
| `03` | 150K | ✅ Confirmed | AIS mic (150K,150K,470K → `03 03 02`) |
| `04` | 56K | ✅ Confirmed | SAT EVF (56K,56K → `04 04`) |
| `05` | 22K | ✅ Confirmed | SAT flash (22K,22K → `05 05`) |
| `06` | 0R/GND | ✅ Confirmed | Dumb Flash (GND,GND,GND → `06 06 06`) |
| `FF` | **N/A** | ✅ Confirmed | **No accessory / Invalid state** (runtime only) → only given in α6600 `libHotShoe.so` |

And we compare that to our given two cameras:

### HX60 Profiles - Correctness

| Profile | Shoe ID | Resistors (ID1, ID2, ID3) | Actions | GPIO Configuration | Your Label | Correct? | Reasoning |
|---------|---------|---------------------------|---------|-------------------|------------|----------|-----------|
| 0 | `05 06 06` | 22K, 0R, 0R | `[0, 1]` | GPIO0=in [0], GPIO1=out [1] | SAT flash variant? / Stereo mic? | ⚠️ Partial / ✅ Correct if stereo-mic | SAT flash is 22K,22K. This has ID2=GND, ID1=GND. Could be a different SAT accessory or a variant. Or as depicted by [@mungewell](https://github.com/mungewell), this could be a stereo microphone - if so, the GPIO configuration (Action `[0, 1]` = GPIO0=input, GPIO1=output) would match the expected behavior for an active audio accessory! |
| 1 | `06 06 06` | 0R, 0R, 0R | `[0, 0]` | GPIO0=in [0], GPIO1=in [0] | Dumb Flash | ✅ Correct | Matches "Replacement MI flash foot" (GND,GND,GND) |
| 2 | `06 06 02` | 0R, 0R, 470K | `[0, 0]` | GPIO0=in [0], GPIO1=in [0] | ADP-MAA | ✅ Correct | Matches "Sony ADP-MAA" (470K,GND,GND) |
| 3 | `03 04 06` | 150K, 56K, 0R | `[1, 0]` | GPIO0=out [1], GPIO1=in [0] | AIS mic variant | ⚠️ Partial | AIS mic is 150K,150K,470K. This has ID2=56K, ID1=GND. Not a perfect match. |
| 4 | `06 02 06` | 0R, 470K, 0R | `[0, 0]` | GPIO0=in [0], GPIO1=in [0] | Unknown | ❓ Unknown | Not in the known table |

### α6700 Profiles - Correctness

| Profile | Shoe ID | Resistors (ID1, ID2, ID3) | Actions | GPIO Configuration | Your Label | Correct? | Reasoning |
|---------|---------|---------------------------|---------|-------------------|------------|----------|-----------|
| 0 | `05 05 02` | 22K, 22K, 470K | `[1]` | GPIO0=out [1] | SAT flash + ADP-MAA hybrid / Stereo-mic derivative | ⚠️ Speculative | SAT flash is 22K,22K (without ID3). The ID3=470K is ADP-MAA-like. No known accessory matches this exact combination. Could also be a stereo microphone derivative - the 22K,22K matches SAT microphone patterns, with 470K on ID3 possibly indicating a different audio configuration or newer microphone variant. The GPIO action `[1]` (output) supports this being an active audio accessory. |
| 1 | `05 05 04` | 22K, 22K, 56K | `[1]` | GPIO0=out [1] | SAT flash + EVF hybrid | ⚠️ Speculative | SAT flash is 22K,22K. SAT EVF is 56K,56K. This is a hybrid not documented anywhere. The GPIO action `[1]` (output) suggests an active accessory. |
| 2 | `06 06 05` | 0R, 0R, 22K | `[0]` | GPIO0=in [0] | Unknown | ❓ Unknown | Not in the known table. GPIO action `[0]` (input) suggests a passive accessory. |
| 3 | `05 06 04` | 22K, 0R, 56K | `[1]` | GPIO0=out [1] | Unknown | ❓ Unknown | Not in the known table. GPIO action `[1]` (output) suggests an active accessory. |

### The Problem

The HX60 and α6700 profiles **don't cleanly map** to the known accessories because. The α6700 profiles appear to represent a **different encoding scheme** or **different accessories** that are not documented in the known table.

mungewell, 2026-09-06:
I trolled through a load of FW images last night and most (if not all) had empty '0x1120005' tables, which doesn't really make much sense. Cameras with hotshoe (and some MI support) must be able to handle devices.

I suspect that this table is **just** for the GPIO, changing the routing etc.... I did not check whether the audio path was adjusted when I was tinkering earlier today. I can also re-assign '06 06 06' and see what happens to the audio.

```
undefined4 FUN_00010eb4(undefined1 param_1, undefined1 param_2, int param_3)
{
    // param_1 = Grp (from descriptor byte 0)
    // param_2 = Bit (from descriptor byte 1)
    // param_3 = action byte
    
    __fd = open("/dev/gpio", 2);
    memset(auStack_28, 0, 0x10);
    local_24 = param_1;  // Grp
    local_20 = param_2;  // Bit
    
    if (param_3 == 0) {
        local_1c = 0x11;  // ← What does 0x11 mean?
    }
    else if (param_3 == 1) {
        local_1c = 9;     // ← What does 9 mean?
    }
    else {
        if (param_3 != 0x10) {
            // Error
        }
        local_1c = 5;     // ← What does 5 mean?
    }
    
    iVar1 = ioctl(__fd, 0xc0104702, auStack_28);
    // 0xc0104702 = GPIO_SET_MODE? 
}
```

BTW 'updatershell' is actually a shell running on the camera, you can issue commands to it's CPU and re-create the above actions.

psiegl, 2026-09-06:
yeah, after my previous analysis I would also claim: this table is merely given for devices acting with GPIO.

mungewell, 2026-09-06:
> Different microphone variants use different GPIO patterns:

Some of the mic's do multichannel analog, and obviously there are digital modes on some mics. Not sure whether the camera (or the ic) clocks the digital data.

psiegl, 2026-09-06:
@mungewell note: I solely obtained this table i.e. 0x1120005 from a running camera. I couldn't retrieve it from a firmware image / software update file. There this table was always empty. From my perspective, this table gets populated during runtime of a camera.

mungewell, 2026-09-06:
> The α6700 profiles appear to represent a different encoding scheme

Did you consider the possibility that a6700 may have the ID sequence backwards? Profile 2 is the mirror of HX-60 profile 0, which is believed to be a stereo mic... obviously the a6700 supports simple analogue mics.

I believe HX-60 profile 3 to be the FDA-EV1MK Viewfinder. Wikipedia's pin-out suggests VFD re-use the audio and flash pins. _They also suggest the the GPS (presumably a UART) uses pin 16._

https://support.d-imaging.sony.co.jp/www/cscs/accessories/compatibility.php?area=gb&lang=en&mdl=FDA-EV1MK&cat=1

mungewell, 2026-09-06:
For the GPIO signals, these are unlikely to be simple 'In or Out'. This is likely a GPIO output controlling a multiplexor on the board, to route the MI signals to one place or another - for example to the audio (MIC) input or from the video system (VFD) differential output.

[image]

A single GPIO could control multiple multiplexors, to switch more that one group of signals. And there may be more complicated routing paths...

[image]

EDIT: maybe more advanced cameras don't require multiplexing, and can programmatically select (for example) internal and external mics from different ADC inputs....

psiegl, 2026-09-07:
> For the GPIO signals, these are unlikely to be simple 'In or Out'. [...]

I think you're right. It is very likely that it won't be just a GPIO that is turned on and off. Got too excited yesterday^^

> Profile 2 is the mirror of HX-60 profile 0 [...]

I mean, Sony changed the SoC with the a6700 (BIONZ XR vs BIONZ X given in a6600; https://en.wikipedia.org/wiki/Bionz) and they also changed the Software Update file (unknown decryption, ...). So was that Sony major overwhole and did they also change this table? Maybe. Sadly the smoking gun is missing. It would be great to obtain the `libHotShoe.so` of the a6700.

My current thought is: is it possible to retrieve the `libHotShoe.so` through `pmca-console.py updatershell` and `bk ...`. I mean, in theory yes I would just need to determine where it is precisely located.

Meanwhile the other option would be to obtain (as you asked for) more tables ( https://github.com/ma1co/Sony-PMCA-RE/issues/735 ), which we can analyse. I was not aware of the japanese Sony support side, that lists compatible equipment. I mean with sufficient table evidences, we could correlate it against the compatibility list.

Another thought I had yesterday:
We know the table now, and we have roughly an idea what entries it requires. With this, one could enhance the table with further entries, with hardware that has its own resistor configuration.

mungewell, 2026-09-07:
Looking at the HX60 FW there isn't a file named that, but something similar.
```
$ find DSCHX60VV210/ -iname "*Shoe.*"
DSCHX60VV210/firmware.tar_unpacked/0700_part_image/dev/nflasha15_unpacked_unpacked/lib/libRsrcHotshoe.so
```

'updatershell' doesn't seem to mount the camera's partitions, but 'serviceshell' does...
```
$ sudo python3 pmca-console.py serviceshell
Using drivers libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Authenticating
Starting service shell...

Welcome to platform shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>pull /usr/lib/libRsrcHotshoe.so
Writing to ./libRsrcHotshoe.so...
>exit
Done

$ ls -al libRsrcHotshoe.so
-rw-r--r-- 1 root root 12548 Sep  7 15:57 libRsrcHotshoe.so
```

You can also use `shell` within 'serviceshell' to navigate and inspect the camera's filesystem.

> one could enhance the table with further entries, with hardware that has its own resistor configuration.

Changing the table didn't make the 'not supported' message go away on the HX-60, there must be another layer in the camera which is accessing MI IDs first. Maybe this is in the Android application layer, rather than Linux.

mungewell, 2026-09-08:
Started working backwards.... I get a message presumably rendered by the Java app, so where does that come from/get triggered???
```
find . -name '*' -exec grep -Hi accessory {} \; 2>&1 | grep binary | sort > accessory.txt 
find . -name '*' -exec grep -Hi supported {} \; 2>&1 | grep binary | sort > supported.txt 
meld supported.txt accessory.txt
```

`nflasha15_unpacked_unpacked/share/app/string_english.uxc` contains string...
```
This accessory is not supported by the
device and cannot be used. Please verify
the compatibility with the device.
```

That file is referenced by `global.xdb`, but I don't know what format this is...
```
hexdump -C ./firmware.tar_unpacked/0700_part_image/dev/nflasha15_unpacked_unpacked/share/app/global.xdb | head
00000000  75 78 61 07 00 00 00 00  09 00 00 00 ca 40 19 00  |uxa..........@..|
00000010  00 00 00 00 2c 00 00 00  50 00 00 00 78 00 00 00  |....,...P...x...|
00000020  9c 00 00 00 b8 00 00 00  e0 00 00 00 00 01 00 00  |................|
00000030  1c 01 00 00 44 01 00 00  70 01 00 00 94 01 00 00  |....D...p.......|
00000040  c0 01 00 00 e8 01 00 00  08 02 00 00 30 02 00 00  |............0...|
00000050  54 02 00 00 70 02 00 00  a4 02 00 00 c8 02 00 00  |T...p...........|
00000060  ec 02 00 00 0c 03 00 00  2c 03 00 00 50 03 00 00  |........,...P...|
00000070  74 03 00 00 94 03 00 00  b4 03 00 00 d4 03 00 00  |t...............|
00000080  f8 03 00 00 18 04 00 00  3c 04 00 00 58 04 00 00  |........<...X...|
00000090  74 04 00 00 94 04 00 00  b8 04 00 00 e0 04 00 00  |t...............|
```

mungewell, 2026-09-08:
From the Android/Java side...
`C0014R.java` contains
```
        public static final int exclametion_image = 2131361881;
```

Which lead me to `com/sony/imaging/app/base/caution/CautionProcessingFunction.java`
```
    private View makeView(int[] cautionId) {
...
                ImageView mImageView = (ImageView) view.findViewById(C0014R.id.exclametion_image);
 
```

I'm  a hardware guy (not programmer) so I don't really see how this is used in the app, I did find something which might be getting coded messages from the underlying (Linux) system, with a bit field '0x2' to signify that a 'caution' window should be used/raised/displayed?
`com/sony/imaging/app/fw/State.java`
```
    boolean _handleMessage(Message msg) {
        if (msg.what != Definition.STATE_TREE_CHANGED) {
            return handleMessage(msg);
        }
        getHandler().removeMessages(Definition.STATE_TREE_CHANGED);
        int handled = msg.arg1;
        if ((handled & 1) == 0 && ApoWrapper.setApoType(getApoType())) {
            handled |= 1;
        }
        if ((handled & 2) == 0) {
            Integer mode = getCautionMode();
            if (mode != null) {
                ((AppRoot) getActivity()).onCautionModeChanged(mode.intValue());
                handled |= 2;
            }
        }
        if ((handled & 4) == 0 && OLEDWrapper.setOledType(getOledType())) {
            handled |= 4;
        }
        if ((handled & 8) == 0) {
            INFO_TYPE mode2 = getHDMIInfoType();
            if (INFO_TYPE.NEUTRAL != mode2) {
                HDMIInfoWrapper.setType(mode2);
                handled |= 8;
            }
        }
        msg.arg1 = handled;
        return handled == 15;
    }
```

psiegl, 2026-09-08:
You are correct with `libRsrcHotshoe.so`, didn't write the full name.

It seems as if the α6700 locked this part down from being read:
```
$ sudo /home/ps/Sony-PMCA-RE/venv/bin/python \
  /home/ps/Sony-PMCA-RE/pmca-console.py serviceshell
Using drivers libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Authenticating
Starting service shell...

Welcome to platform shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>pull /usr/lib/libRsrcHotshoe.so
Writing to ./libRsrcHotshoe.so...
Error: Senser file control error 128
>
```
while `shell` seems broken or not properly working:
```
>shell
Terminal activated. Press <CTRL+C> to exit.
ls
Error: 
>Please press enter

```

psiegl, 2026-09-08:
So, I bought the [DJI Mic Series Camera Adapter](https://store.dji.com/de/product/dji-mic-series-camera-adapter?set_region=DE&vid=182691), which is recognised by the α6700 as a "mi Schuh Audio" with options (48kHz/24bit 4ch; 48kHz/24bit 2ch;48kHz/16bit 2ch):  
[image]

However, the device is solely recognised, if I put the camera in video capture mode. It is not power up the small PCB attached when using camera or time lapse mode. Furthermore I noticed, that the little Nordic Semi on the PCB + a bit more attached is too much for the electric current given. Which seems odd, as Pin 17 should supply 1.7A with 6.1V-8.4V ( https://en.wikipedia.org/wiki/Multi_Interface_Shoe ), which even with a simple buck converter for the USB-C port should suffice.

I tried already [DJI Mic 2 Camera Adapter](https://store.dji.com/de/product/dji-mic-2-camera-adapter?set_region=DE&vid=163881), however that one was not recognised at all (maybe broken?).

Next step is to dissassemble the the 1st one and trying to figure out what resistors it employs.

psiegl, 2026-09-09:
Well, if there is no spec, ... a testbench might be useful :)

[image]

First attempt: reproduce the profile 0 of the α6700:

| Profile | Shoe ID | Resistors (ID1, ID2, ID3) |
|---------|---------|---------------------------|
| 0 | `05 05 02` | 22K, 22K, 470K |

And .. well .. it is the stereo mic (see the 48kHz/16bit 2ch).

[image]
Normally, the menu is also grayed out, if no MI adapter is present however:

[image]

Quite a success ... now it is just a matter of time and effort to go through the 6³ configurations (well first the known ones).

mungewell, 2026-09-10:
> Quite a success

Yes, congratulations. Did you have to connect pins 21 and/or 24 (part of the hot-shoe 'foot')

psiegl, 2026-09-10:
To determine if UNREG (Pin 17) in combination with PGND (Pin 21) and/or LOGIC_VDD (Pin 13) in combination with any REG_GND (currently PIN 7) supplies power, the - new - setup looks as follows:

[image]

psiegl, 2026-09-10:
A first attempt to reverse engineer through trying out the combinations... I focused on the known profiles and adapters given:

[image]

psiegl, 2026-09-10:
since the Alpha 6700 is not rejecting profiles of the HX60 immediately with a flash screen telling me so, well ... if we consider below again:

```
=== RX100M4 ===
Action Count (P): 0
Profile Count (N): 0
Descriptors: []
Profiles:

=== a7R II ===
Action Count (P): 0
Profile Count (N): 0
Descriptors: []
Profiles:

=== HX60 ===
Action Count (P): 2
Profile Count (N): 5
Descriptors: ['080000', '021100']
Profiles:
  [0] Shoe ID: 050606, Actions: [0, 1]
  [1] Shoe ID: 060606, Actions: [0, 0]
  [2] Shoe ID: 060602, Actions: [0, 0]
  [3] Shoe ID: 030406, Actions: [1, 0]
  [4] Shoe ID: 060206, Actions: [0, 0]

=== α6700 ===
Action Count (P): 1
Profile Count (N): 4
Descriptors: ['030800']
Profiles:
  [0] Shoe ID: 050502, Actions: [1]
  [1] Shoe ID: 050504, Actions: [1]
  [2] Shoe ID: 060605, Actions: [0]
  [3] Shoe ID: 050604, Actions: [1]
```

... I have doubts, but just an idea... could the descriptor be a mask?

psiegl, 2026-09-12:
I went through all 6³ configurations and obtained:
[Alpha6700 id combinations.ods](https://github.com/user-attachments/files/32149076/Alpha6700.id.combinations.ods)
As the camera was very clear on telling if likely an analog audio device is given, that should be set to digital. Or just told me it is a digital audio device, the associated ID combinations was easily to identify. As the flash likely requires more (see comment above), I would likely need to bridge likely F*-Pins, I wasn't able to detect within the camera menu a specific flash support. Even though some of the profiles will be flashes.

For an Alpha6700 the most interesting profile is 050606, as it enables power in all configurations, is - not - directly supported, as considered as analog audio. Thus, if dial is in movie, it does not disable its internal mic which would be unhandy if an own shoe adapter is developed, that is not a mic.

mungewell, 2026-09-13:
> I went through all 6³ configurations

Wow! A lot of legwork but that information is really helpful to the DIY community, and hopefully (especially) useful in your own project.

From my project's point of view it is interested to note:
1). "050606" seems to be the analog mic to target. I know that there are others which do a center channel, perhaps one of the other analog codes is the audio out??

2) The a6700 can ONLY do digital via the hot-shoe. Maybe not too much of a surprise, but would mean that a PicoTimecode would have to support digital audio if we wanted to use the hot-shoe on some cameras. The CPU is not involved in the precise LTC output, and we have use 100% of the rp2040's PIO - maybe we need to shift to rp2350 which has an additional PIO block (4x state-machines) to emulate an I2S device - guessing that's the protocol.

3) I guess the instruction "switch to digital" is because the dual mode devices change which resistors they present, so that they still function on older (analog only) models. 

For your comment on power, did you see '050604'? 

Does the power turn off when it the camera is turned 'off', but still with battery connected? For PicoTimecode we want power all the time, as we don't want the LTC clock to be interrupted/reset.

psiegl, 2026-09-13:
> For your comment on power, did you see '050604'?
> 
> Does the power turn off when it the camera is turned 'off', but still with battery connected? For PicoTimecode we want power all the > time, as we don't want the LTC clock to be interrupted/reset.

Well yes:
[image]

The moment, the camera (Alpha6700) is turned off, the power given by UNREG is gone. :-(

mungewell, 2026-09-17:
@psiegl My Digikey order is in, sorry I haven't been more pro-active on this.... hyper focused on bug hunting on STC, and getting the new release out.

Should be able to confirm behavior some time next week.

psiegl, 2026-09-20:
Debugging and more debugging :-)

I found the following for an Alpha6300:
[image]
on page 28 of the [ILCE-6300_6300L_6300M Service Manual Level 3.pdf](https://github.com/user-attachments/files/32444561/ILCE-6300_6300L_6300M.Service.Manual.Level.3.pdf).

Same for the Alpha6000:
[image]
on page 26 of the [ILCE-6000 Service Manual Level 3](https://it.ifixit.com/Document/YmcODnMHSUP4xj1f/sony_ilce-6000_ilce-6000l_ilce-6000y_ver.1.1_rev-1_level3.pdf).

And the Alpha6500:
[image]
on page 32 of the [ILCE-6500 Service Manual Level 3](https://www.manualslib.com/manual/2999481/Sony-Ilce-6500.html).

In contrast on the RX100M2, the power supply is 2.8V, however the same resistors internally:
[image]
on page 18 of the [DSC-RX100M2 Service Manual Level 3](https://elektrotanya.com/sony_dsc-rx100m2_ver.1.0_level3.pdf/download.html).

Thus, for each ID pin, we have the following schematic:
```
A_3.15/2.8V_MSS  ──[ 100 kΩ ]──┬── ADC1_INx
                               └── SHOE_IDx ──[ Rx ]── GND
```

I gave this information also to Deepseek AI and provided it again with the - full - Ghidra output ( [alpha6600_libRsrcHotshoe.so_ghidra.txt](https://github.com/user-attachments/files/32444940/alpha6600_libRsrcHotShoe.so_ghidra.txt) ) of an Alpha6600 `libRsrcHotshoe.so`, i.e. not just C/C++ generated by Ghidra. I also told, that so far ID to GND, 22 kOhm, 56 kOhm, 150 kOhm, 470 kOhm and floating seemed fine, but I am still unsure if these values are right?! I mean, so far they either come from reverse-engineerd Sony Adapters or some people trying (including me :) ). On reason for my suspision: I am measuring the reistors on the [DJI Mic 2 Camera Adapter](https://store.dji.com/de/product/dji-mic-2-camera-adapter?vid=163881) and I am not so much happy ... (you need to zoom in to the left). Still, the lower left two resistors should (if I have followed the FFC cable correctly) handle the IDs, at least they seem to be the closest to the ID pins.
[image]

Nevertheless, with the full Ghidra output and the hint on the ID wiring of the RX100M2 and the Alphas, I obtained (which sounds reasonable):

> ## MI Shoe ID detection — analysis of `libRsrcHotshoe.so`
>
> ### 1. ADC threshold table is populated at runtime
>
> `FUN_000111f8` (called from both `rsrc_hotshoe_init` and `rsrc_hotshoe_res`) fills a six-entry 16-bit threshold table in `.bss` at `DAT_0001b1fe`–`DAT_0001b208`. Each entry is first attempted from a per-chassis NVRAM backup key, and only falls back to a compiled-in default if that read fails:
>
> ```asm
> 00011286  ldr   r4,[DAT_000114c0]   ; base pointer
> 00011288  ldr   r0,[DAT_00011488]   ; = 016C0003h  (backup key)
> 0001128a  add   r4,pc
> 0001128c  mov   r2,r4
> 0001128e  bl    FUN_000111a0        ; read 2 bytes from NVRAM into [r4]
> 00011292  cbz   r0,LAB_0001129a
> 00011294  mov.w r3,#0x388
> 00011298  strh  r3,[r4,#0x0]        ; fallback default = 0x0388
> ```
>
> The same pattern repeats with keys `016C0004`–`016C0008` and defaults `0x02C8`, `0x01FC`, `0x0134`, `0x0044`, `0x0000`, writing consecutively at `[r4+2]`, `[r4+4]`, …, `[r4+0xA]`. So the table in memory (ascending address) is:
>
> ```
> DAT_0001b1fe : 0x0388  (904)
> DAT_0001b200 : 0x02C8  (712)
> DAT_0001b202 : 0x01FC  (508)
> DAT_0001b204 : 0x0134  (308)
> DAT_0001b206 : 0x0044  (68)
> DAT_0001b208 : 0x0000  (0)
> ```
>
> ### 2. Comparison logic in `RsrcHotshoeGetShoeID`
>
> ```asm
> 00011c48  subs  r2,r3,#0x2        ; r3 = &DAT_0001b1fe ; r2 = &DAT_0001b1fc
> 00011c4a  movs  r3,#0x0           ; r3 = code counter = 0
> 00011c4c  ldrh.w r5,[r2,#0x2]!    ; pre-increment: r2 += 2 ; r5 = *r2 (first = 0x0388)
> 00011c50  add.w r3,r3,#0x1        ; r3 = index+1
> 00011c54  cmp   r5,r1             ; compare threshold with ADC value
> 00011c56  bhi   LAB_00011c5c       ; branch if threshold > adc → keep scanning
> 00011c58  uxtb  r3,r3             ; else r3 is the final code
> 00011c5a  b     LAB_00011c62
> 00011c5c  cmp   r3,#0x6
> 00011c5e  bne   LAB_00011c4c       ; not yet at index 6 → next threshold
[... 78 lines cut ...]
>
> Effective resistor windows (for building an injection device — one resistor per ID pin, camera's own 100 kΩ pull-up does the rest):
>
> | Code | Rx range | Nominal |
> |---|---|---|
> | 1 | > 756 kΩ | open |
> | 2 | 365 k – 756 k | 470 kΩ |
> | 3 | 121 k – 365 k | 150 kΩ |
> | 4 | 38 k – 121 k | 56 kΩ |
> | 5 | 7.1 k – 38 k | 22 kΩ |
> | 6 | 0 – 7.1 k | 0 Ω |
>
> Windows are ~3× wide, so 5% tolerance parts are adequate. Watch the edges: keep code-2 resistors well above 365 kΩ, and never use anything below 7.1 kΩ where code 5 is intended — that crosses into the code-6 bin.
>
> ### 4. Caveats
>
> Three things the schematic plus firmware dump do not fully pin down:
>
> 1. **NVRAM overrides.** `FUN_000111f8` prefers the per-chassis values stored at backup keys `016C0003`–`016C0008` and only uses the `.text` immediates as fallback. On production units the actual bin edges may differ slightly from the defaults. This matters only near a bin boundary; mid-bin values like the nominals above are unaffected.
>
> 2. **Pull-up value per chassis.** The Rx-boundary table above assumes `R_pullup = 100 kΩ`, which the α-series schematic confirms. If a different chassis uses a different pull-up value, the *counts* per resistor shift, and while the codes remain monotonic, the specific ohm values per bin would need to be re-derived from that chassis's schematic. The per-chassis NVRAM thresholds are the mechanism that compensates for this, but the  [... cut ...] r than a rail-derived one, then the voltages would need to be interpreted against that fixed reference and the code-vs-resistance mapping would shift with rail voltage. The α-series and RX100M2 evidence is consistent with ratiometric operation, but this is worth verifying per chassis if the rail differs significantly.

Furthermore, I dumped the same Ghidra output (not C/C++) from an ILCE7M2 and HX60V into Deepseek, trying to determine if at least the initial table is the same -> that's the case.

psiegl, 2026-09-20:
So the wiring of 3.15V and 100 kOhm Pull-Ups was already given for Smart Accessory Terminal, see IHS_ID1 and IHS_ID2:
[image]

mungewell, 2026-09-22:
Success on the HX-60, with PicoTimecode (rev3 board) and the "flex + hot-shoe" from SMAD-P3!

Initially I had ID1 and ID3 swapped (got 'not compatible' message), but with that fixed I was able to record video with audio-timecode via the hotshoe.
[image]

There's some for of AGC, so it didn't mind Mic or Line level. Audio not be best quality, but it did decode.
[image]

```
$ ./ltcdump -c 1 ~/Music/MAH09261.wav 
Note: This is not a mono audio file - using channel 1
#User bits  Timecode   |    Pos. (samples)
#DISCONTINUITY
5049434f   01:00:03:23 |      451     2362  
#DISCONTINUITY
5049434f   01:00:03:21 |     2363     4282  
5049434f   01:00:03:22 |     4283     6202  
5049434f   01:00:03:23 |     6203     8123  
5049434f   01:00:03:24 |     8124    10042  
5049434f   01:00:04:00 |    10043    11962  
5049434f   01:00:04:01 |    11963    13880  
5049434f   01:00:04:02 |    13881    15799
...
```

mungewell, 2026-09-23:
From the service manuals; Remember when we were talking about the `0x01120005` table, and speculated what the two GPIOs might be doing....

[image]

So the RX100 uses USB2.x switch ICs to route either internal or external MIC to the processor. Probably same on the HX-60 and others.

Note: signal is actually analogue, the USB2.x switch is just a FET switch IC.

mungewell, 2026-09-23:
Re-did the audio connections; with correct MGROUND and 1.0uF caps instead of 0R as service manual from RX100m3 suggested that there's a DC bias for the Mic.

Audio still a bit crappy, but at-least it's balanced L/R now.
[image]

Video clip of it in action:
https://youtu.be/nxgzcQfsGgY

psiegl, 2026-09-24:
@mungewell 
> If we had the table from the RX100mk2, we could trace to particular pins on the CPU.

So, I had the idea of obtaining one Sony with MI slot that I can get as cheap as possible.
Luckily, I ended up with a 'broken' (it seems fixable .. let's see ... ) RX100M2, which at least gave me the table:

https://github.com/ma1co/Sony-PMCA-RE/issues/735#issuecomment-5817151190

```
Using drivers libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to service mode

Waiting for camera to switch...
Found a camera in service mode
Authenticating
Starting service shell...

Welcome to platform shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>bk r 0x01120005
42 49 4e 21 01 04 08 03 00 05 06 06 00 06 06 06 00 06 06 02 00 03 04 06 01 46 49 4e 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
>info
Model:              DSC-RX100M2
Product code:       80814150
Serial number:      3021491
Backup region:      CX63711_CE3
Firmware version:   1.10
```

:-)

mungewell, 2026-09-24:
The full decode of the rx100mk2's table is
```
Container: 
    num_of_gpio = 1
    num_of_id = 4
    gpio = ListContainer: 
        b'\x08\x03\x00' (total 3)
    ids = ListContainer: 
        Container: 
            id = b'\x05\x06\x06' (total 3)
            use_gpio = ListContainer: 
                0
        Container: 
            id = b'\x06\x06\x06' (total 3)
            use_gpio = ListContainer: 
                0
        Container: 
            id = b'\x06\x06\x02' (total 3)
            use_gpio = ListContainer: 
                0
        Container: 
            id = b'\x03\x04\x06' (total 3)
            use_gpio = ListContainer: 
                1

```

So it contains the same ID as HX60 (id = b'\x03\x04\x06', believed to be ViewFinder) and 'uses' that one GPIO (b'\x08\x03\x00') when that is attached...

[image]

In the datasheet for the CPU there will be a big table which defines the pin-mode for each pin. Modern CPUs usually have 8 (or so) functions for each pin, I suspect that one (or more) of these will be for "Y3 = GPIOxx"... which might match the table in some form.

Good luck mending your new acquisition.