mungewell/pico-timecode #18: Timecode over USB-HID

mungewell, 2024-12-04:
FujiFilm announced a FW upgrade to introduce compatibility with Ambient Lockit devices, talked about/demonstrated here:
https://youtu.be/C4O3qpItzw0?si=_slQyaDvqQGjLbxZ

Big question is whether this is an Open Standard, and whether 'we' could implement with PicoTimecode.

Micropython on the RP2040 now supports implementing USB devices (apparently):
https://github.com/micropython/micropython/blob/master/examples/usb/usb_simple_device.py

flummer, 2024-12-04:
I think it is MIDI time code that is used

https://ambient.de/en/products/acn-cl-lockit-timecode-und-syncgenerator (technical details tab)

mungewell, 2024-12-06:
@flummer Thanks for the link, I first saw mention of this in Instagram post - where they seem to say that you can use either USB-HID or MTC. Obviously USB-Midi Class and USB-HID are different things... I know you can convert MTC to TC (afterall my Sync-IO can do that), but I'm not sure how simple it would be for a Pico.
https://www.instagram.com/p/DCpGFQ6sQTD/?igsh=YXhvam1iaGRnYml1

I did find that there is a 'HID Usage' for a 'Time Sensor', and perhaps this is what they use. I'm not clear how one would relate fractional framerates and/or dropframe into that....
[image]

mungewell, 2025-11-21:
Doubled checked that the midi stuff does work, simple script which fakes sending MTC (I think...)
[mtc_example.py.zip](https://github.com/user-attachments/files/23667430/mtc_example.py.zip)

Need to install module to Pico from PC console.
```
$ mpremote mip install usb-device-midi
Install usb-device-midi
Installing usb-device-midi (latest) from https://micropython.org/pi/v2 to /lib
Installing: /lib/usb/device/__init__.mpy
Installing: /lib/usb/device/core.mpy
Installing: /lib/usb/device/midi.mpy        
Done
```

mungewell, 2025-11-22:
Improved test script (don't forget to install the 'usb-device-midi' lib):
[mtc_example.py.txt](https://github.com/user-attachments/files/23686193/mtc_example.py.txt)

Here it is sending MTC to 'xjadeo' controlling the playback head. Counts to '00:02:00:00' and then restarts.
[image]

_Next up porting into project._

mungewell, 2025-11-22:
Prototype code running MTC output from Pico-Timecode is here:
https://github.com/mungewell/pico-timecode/tree/mtc_output

Demo video:
https://www.youtube.com/watch?v=kkTFQ0azi5w